meshtastic/network-management-client #158: Personal Goals

samfydega, 2023-01-04:
# Sam Crombie | Personal Goals
*what does success for this class mean to you?*

* I get efficient at full-stack development
* Finally understanding Typescript and Redux so that I can use it in other projects
* We get to conduct a field test of the software with the NH volunteer rescue team we've spoken with 
* Our contribution continues to be worked on by the Meshtastic team
* I earn the respect and admiration of Tim and Natalie

## Challenges
*what are some challenges you foresee?*

* Scoping our sprints to achieve enough progress each week
* Getting the product to a point where we are comfortable conducting a field test with a volunteer team
* Getting the client stable and compatible across platforms

ajmcquilkin, 2023-01-08:
@samfydega This should be on the fork lol

nataliesvoboda, 2023-01-10:
Hi Sam, A field test - wow yes that would be wonderful. I wonder if you could break this down into smaller goals in case the "high bar" doesn't work out. Could this test be just on campus with your team?

closed 2023-01-23 by ajmcquilkin