meshtastic/network-management-client #160: Build of rust part is broken in master branch

D4rk4, 2023-01-08:
```
error: failed to run custom build command for `app v0.1.0 (/home/dark/repos/Meshtastic-emergency-response-client/src-tauri)`

Caused by:
  process didn't exit successfully: `/home/dark/repos/Meshtastic-emergency-response-client/src-tauri/target/debug/build/app-e84d1d451526115d/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=protobufs
  Running: "/home/dark/.cargo/registry/src/github.com-1ecc6299db9ec823/protoc-bin-vendored-linux-x86_64-3.0.0/bin/protoc" "--include_imports" "--include_source_info" "-o" "/tmp/prost-buildXLPArR/prost-descriptor-set" "-I" "protobufs"

  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: Other, error: "protoc failed: Missing input file.\n" }', build.rs:37:54
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...

🚀 A new version of Tauri CLI is available! [1.2.2]
 ELIFECYCLE  Command failed with exit code 101.

```

ajmcquilkin, 2023-01-08:
Hi D4rk4! Have you initialized the `protobufs` git submodule? I would try running `git submodule update --init` if not. It looks like the `build.rs` script can't find the `.proto` files it's trying to build

D4rk4, 2023-01-09:
@ajmcquilkin you're right I forgot to init git submodules. Please add this to the readme.

closed 2023-01-09 by D4rk4

ajmcquilkin, 2023-01-09:
Glad to hear it! Where would you be looking to see it? I think we might have mentioned it in the `Prerequisites` section but I'd be happy to move it if you think there's a more intuitive place to put it

D4rk4, 2023-01-09:
Oh sorry. I guess better to move this near to `pnpm install`

ajmcquilkin, 2023-01-09:
No worries! I'll move it into the command section, a lot of people miss that command :)