meshtastic/network-management-client #182: [UI] Add UI for visualizing node connections

ajmcquilkin, 2023-01-09:
Design UI to allow the user to view previous "connections" between neighboring nodes (i.e., nodes that a given device has previously received packets from). Current thinking goes that this UI should only be visible when the given node is selected, to reduce complexity on the map. This UI should include visual representations of the strength of each connection (think SNR).

ajmcquilkin, 2023-01-24:
Hey GUVWAF, thanks for the thoughts!

This ticket specifically is going to be using simulated data for the time being so we can get a proof of concept out for the UI, but we're envisioning the graph-building functionality following the [Discourse discussion](https://meshtastic.discourse.group/t/enabling-network-analysis-for-meshtastic-networks/7063/8). We haven't built out any functionality that would directly query the network for graph-building information, but we'll either be extending the Traceroute module or continuing to analyze the impact of the packet proposed as part of that discussion.

I'm also going to get @barkincavdaroglu and @uhuruhashimoto in on this conversation as well, as they've done more specific work with the packet handling.

barkincavdaroglu, 2023-01-24:
We had a similar conversation, and ending up with a star graph was one of my main concerns too. Currently all the algorithms assume we have the whole network topology. Our current plan is (correct me if I'm wrong @uhuruhashimoto) is to construct 1-hop neighborhood for each node and then aggregate these neighborhoods into centralized graph.

uhuruhashimoto, 2023-01-24:
Tagging in to the conversation -- great summary @barkincavdaroglu!

To recap, we considered extending Traceroute, but are leaning away from it, because of the network traffic hit. I believe we would have to run `n^2` TraceRoute operations every `x` amount of time to get a picture of the graph, and the docs note that the operation is slow. The nature of adding a new packet to allow each node to broadcast its own info would mean that we would gain the same information without the Traceroute overhead. (@ajmcquilkin also has a discussion of this in the [Discourse](https://meshtastic.discourse.group/t/enabling-network-analysis-for-meshtastic-networks/7063/8).)

Our plan for the future is to (1) simulate the effects of our new packet on the network (ideally with an explicit comparison with Traceroute) and (2) mock network topology info to allow us to set up and test the frontend (which is the assumption this ticket rests on). 

@GUVWAF - Curious if you have any thoughts on this plan?

GUVWAF, 2023-01-24:
@ajmcquilkin Ah, I was moving a bit too fast then ;) Excited to see where this is going.
@barkincavdaroglu @uhuruhashimoto Aggregating the 1-hop (direct connections, I would actually call it 0-hop) neighborhoods would work as well. However, since a packet that is flooded by an intermediate node is an exact copy (except that the hop limit is bumped down by one) of the original packet, you cannot know what are the direct connections of a node using the info of normal traffic. I think the options you have are: 

1. Use the Traceroute, which is a rather small packet compared to the proposed new packet, but indeed should be sent to each individual node. However, if you do this more clever and start with nodes that are multiple hops away, you can derive the path to an intermediate node as well. And if you assume an undirected graph, you would have to do the Traceroute only in one direction. 
As I said, you could add the SNR per hop to this packet. 
2. Get the hop limit setting of each node, such that you know there is a direct connection if you receive a packet that still has this original hop limit. Then you can aggregate the 0-hop neighborhoods to form the graph, but that leads to overhead as well.
3. Introduce a special packet that is sent with a hop limit of 0 to derive the 0-hop neighborhoods and aggregate them to form the graph, like in 2.

GUVWAF, 2023-02-15:
If you're going to simulate the effects of your new packet, make sure to merge [this PR](https://github.com/meshtastic/firmware/pull/2272) such that you can get channel utilization metrics using the interactive simulator, like this: 
[image]

BTW, it came to my mind that you can also add a broadcast Traceroute option. Then only nodes that do not hear a rebroadcast from another node need to answer the request, because they are at the edge of the network. 
One thing is that since the protocol prefers to take the route with least hops, it might be that you ‘skip’ nodes as seen here: 
[image]

Either you can then assume that nodes that are not on a path have a direct connection to the most nearby nodes, or do an additional Traceroute request to those specific nodes.

GUVWAF, 2023-02-17:
@uhuruhashimoto Nice to hear you’re planning to work on this even after your deadlines!

I think you've correctly identified the problems and I agree that the `TraceRoute` might not be the optimal solution. BTW, the best guess for the hop limit would be `HOP_RELIABLE`, which is the default (equal to 3), but indeed you can't be sure about that.

I really like the idea of the `LastSentByID` in the header, especially since these two bytes (I assume you pick 2 out of the 4 bytes of the ID?) are now unused. Actually when Meshtastic has decided on making a breaking change, I want to propose cutting the NodeID to 2 bytes and create a new header with `originalTxID`, `currentTxID` (which is your `LastSentByID`), `nextHopID` and `destinationID`. The `nextHopID` is a neighbor you think is the best next hop in the route to the destination. I think these are the necessary ingredients to make a more clever routing algorithm in which a specific route to a certain destination can be set up, which I would like to eventually see within Meshtastic.

Your approach seems like a clever way to discover node neighbors without adding any overhead, as opposed to `TraceRoute`. However, it is not completely clear to me when the `NeighborInfo` will be sent. Do I understand correctly from this: 

> Wrapping the NeighborInfo packet as a MeshPacket payload with the associated metadata (for every single packet) is worse than a perfect-solution TraceRoute packet, which just appends the node numbers to its payload directly from the router. Also, if nodes want to opt out, the extra field in the MeshPacket just ends up costing us bandwidth.

...that you will be sending this with every single packet? If so, indeed I'm afraid it will not end up in the firmware due to the downsides you mention. I think a dedicated Module has to be made for it, for which users can opt-in and configure a broadcast interval. But I totally understand that you're going for this approach when testing and it can be changed later.

Also, I recently made [some optimizations](https://github.com/meshtastic/firmware/pull/2276) to the firmware, which prevents a storm of messages when sending the NodeInfo when all nodes are started up at the same time. It would be best to merge this in as well, otherwise you will be waiting for a long time before the “_NodeInfo storm_” is over when simulating ;) 
Please feel free to ask me any questions regarding the simulator. It has mainly been used (by me) for development, so it may not always be very user-friendly.

uhuruhashimoto, 2023-02-22:
Thanks for the feedback! 

There will be a dedicated module for the NeighborInfo module -- there's one up on an experimental branch on a fork I have [here](https://github.com/uhuruhashimoto/firmware/pull/5). We're planning on following the NodeInfo model and allowing users to opt-in and, if they do, set the frequency of updates. 

As for the packet header -- that's a cool idea! I was looking at it today and was also thinking about compressing the packetID to get 4 bytes of lastSentByID in to the header (since it's sequential after a random init). I have two questions related to reducing node number size: 
1. Collisions. Are you worried at all about the birthday paradox when reducing node number sizes? With 2 bytes, we'll get >50% chance of collisions at 256 nodes, and less than a 1% chance of collisions until 37 nodes. But if our nodes change their IDs periodically or something, maybe that will become a problem.
2. Do we need 4 bytes of randomness for any security reason? That doesn't seem like much to brute force, and seems irrelevant given that the system security will be relying on symmetric AES keys anyway.

I'll pull the updated firmware and test it! I've been definitely watching the NodeInfo storm happen :) 

The simulator is proving very helpful! I have one error that I'll raise on the repo - when I initialize a not fully-connected graph (for example, three nodes in a row), I sometimes get it to crash with a math error. Also, I believe we're seeding the random generator with the same value to generate node IDs (they're always ending up as 16 and up on my machine); is that intentional?

uhuruhashimoto, 2023-02-22:
Also, do you have any ideas on what routing algorithm to use? Are you going to use the GPS info we have in the NodeDB to help determine which nodes might be most likely (you could even take the smallest GPS difference between target node and all other nodes, or use the elevation field to determine repeaters and give them priority... so many options!) I looked into routing algorithms a bit in the fall and there were some really cool ones out there.

GUVWAF, 2023-02-22:
Sounds good! I will have a closer look at the NeighborInfo module later.

1. I’m not too worried about the birthday paradox, since the devices can only hold about 80 nodes in the NodeDB due to limited RAM. In practice I think most meshes will not exceed 20 nodes. Besides that, I think a mechanism can be put into place to detect and correct ID collisions. 
2. I don’t think the 4 bytes are needed for security. The cryptographic nonce is currently made up of the packetID and ‘fromNode’, but I guess that can be combined with e.g. `destinationID` and `nextHopID`.

Don’t hesitate to raise the issue on the simulator. 
Indeed, the simulator will generate IDs as 16 and up. I did this, such that I do not first have to map the ID to the specific interface, but this can be changed. It starts at 16 because IDs below 4 are reserved and 16 is 0x10 in hex, such that you can see at a glance in the logs which node it was.

Right now I think a distance-vector routing protocol (like Routing Information Protocol) is most suited. I’m not sure if using GPS info is a good idea, since not all nodes have a GPS and generally SNR information would be more helpful in an RF environment.

PR #390 (2023-05-19): Integrate deck.gl
[image]

This PR updates the map UI to use the [deck.gl](https://deck.gl/) map and graphics library. The deck.gl library is highly optimized for graphics overlays over mapping, which this PR uses to display node connections and locations.

Additionally, this PR adds node and edge tooltips, and updates many icons to use the [Lucide Icons](https://lucide.dev/) library.

Closes #77 
Closes #92 
Closes #182
Closes #315
Closes #364 
Closes #365
closed 2023-05-19 by ajmcquilkin