meshtastic/network-management-client #262: Broken dependency in rust?

D4rk4, 2023-02-06:
```
pnpm run rust:dev

> meshtastic-emergency-response-client@0.0.0 rust:dev /home/dark/repos/Meshtastic-emergency-response-client
> tauri dev

     Running BeforeDevCommand (`pnpm run ui:dev`)

> meshtastic-emergency-response-client@0.0.0 ui:dev /home/dark/repos/Meshtastic-emergency-response-client
> vite

  VITE v4.0.4  ready in 751 ms

  ➜  Local:   http://localhost:5173/
  ➜  Network: use --host to expose
        Info Watching /home/dark/repos/Meshtastic-emergency-response-client/src-tauri for changes...
    Updating crates.io index
error: failed to select a version for the requirement `nalgebra = "^0.32.1"`
candidate versions found which didn't match: 0.31.4, 0.31.3, 0.31.2, ...
location searched: crates.io index
required by package `app v0.1.0 (/home/dark/repos/Meshtastic-emergency-response-client/src-tauri)`
🚀 A new version of Tauri CLI is available! [1.2.3]
 ELIFECYCLE  Command failed with exit code 101.

```

ajmcquilkin, 2023-02-06:
Interesting, I've not seen this error. Which OS are you running? And just for And one thing to try might be to wipe the `/src-tauri/target` directory and rebuild with `pnpm run rust:dev` again

D4rk4, 2023-02-06:
Kali Roling/Debian mix,
Wipe of "/src-tauri/target" did not help

ajmcquilkin, 2023-02-06:
Hmmm okay, I can look into that a bit later in the week. I just checked the `Cargo.toml` file and `0.32.1` is the intended version of the `nalgebra` package we're using. I'm wondering if this could be an issue with how your computer is acquiring the `crates.io` index that the error message references?

D4rk4, 2023-02-06:
>  I'm wondering if this could be an issue with how your computer is acquiring the crates.io index that the error message references?

How I can check/debug this?

ajmcquilkin, 2023-02-06:
Did some digging, apparently this is an issue with Cargo's cache becoming corrupted. The links I found either recommended updating your Rust version via `rustup` or by clearing certain directories within the Rust toolchain. I've attached some links below.

1. https://substrate.stackexchange.com/questions/3628/failed-to-select-a-version-for-the-requirement-wasmtime
2. https://blog.illixion.com/2021/10/fix-failed-to-select-a-version-cargo/

D4rk4, 2023-02-06:
thx @ajmcquilkin removing of `~/.cargo/registry` resolved the issue. Maybe add this to readme: 

> If you expecting issues with building of rust part ... remove `~/.cargo/registry`

closed 2023-02-06 by D4rk4

ajmcquilkin, 2023-02-06:
Glad to hear it! I spoke with the team and I'm not sure this was related to our project specifically, I think this was an issue with the Cargo project. Happy to add it if this becomes a common issue