meshtastic/network-management-client #299: Add ability to define meshtstic port via command line option

D4rk4, 2023-02-20:
Nice to have the ability to define mehtastic device port via the command line option.
Why?
- Application used by not technically trained user => sysop may define port in shortcut/shell script.
- Application running automatically when OS booted up and ready to action (kiosk mode)

ajmcquilkin, 2023-02-20:
That's an interesting idea, I'd love to see this!

What do you think should happen if a CLI argument was used to launch the application (maybe `--port SOME_PORT`) and connection on that port failed? Would you want the application to start normally or would you want the launch to fail?

D4rk4, 2023-02-20:
Better for UX just show a warning, and continue.

ajmcquilkin, 2023-03-01:
Update on this -- I've tried to tackle this ticket a couple of times, but it seems like Tauri is failing to parse CLI arguments when in development mode. Might need to hold off on this ticket for the time being, although this is absolutely something I want to get into the application. Also happy to accept contributions on this if anyone has experience with running Tauri with CLI args. 

This is the guide I've been working with: https://tauri.app/v1/guides/features/cli/

ajmcquilkin, 2023-03-23:
Update on this issue -- the solution I've found is to only test the CLI using the `pnpm run rust:build` command then running the built executable from there (`EXECUTABLE_PATH --port PORT_LOCATION`). There's still some UI issues, but there's functional code on the [cli-port-connection](https://github.com/ajmcquilkin/meshtastic-emergency-response-client/tree/ajmcquilkin/cli-port-connection) branch.

closed 2023-04-05 by ajmcquilkin

D4rk4, 2023-05-02:
When run with --port COM4, I see port pick dialog, butin logs I observe communication messages with MCU. UI in master still not handle it?

ajmcquilkin, 2023-05-02:
Let me push an update to the CLI features, I think a regression snuck into the code.

ajmcquilkin, 2023-05-02:
Just pushed an update that 1) allows for running the CLI arguments with the `pnpm run rust:dev` command, and 2) that removes an error within the UI code that occasionally caused the UI to miss the "device configured" event

D4rk4, 2023-05-02:
Working!

PR #357 (2023-04-05): Add CLI option to programatically connect to devices on startup
Per suggestion by @d4rk4, this PR adds in the ability to programmatically connect to Meshtastic serial devices instead of manually selecting them during application launch. The syntax to do so follows the pattern `EXECUTABLE --port PORT_NAME` or `EXECUTABLE -P PORT_NAME`. If a port is specified, the application will store its name in Tauri state.

The UI layer then queries if the application has a port marked for automatic connection (the `request_autoconnect_port` command). If this command returns a valid port name, the UI layer  [... cut]