meshtastic/network-management-client #352: Broken submodule defs

D4rk4, 2023-03-07:
```
dark@hope{~/repos/meshtastic-emergency-response-client}:cat .gitmodules 
[submodule "src-tauri/protobufs"]
	path = src-tauri/protobufs
	url = https://github.com/uhuruhashimoto/protobufs.git
	branch = neighborinfo_module
dark@hope{~/repos/meshtastic-emergency-response-client}:git submodule update --init
fatal: No url found for submodule path 'protobufs' in .gitmodules
```

ajmcquilkin, 2023-03-12:
Sorry, forgot to update this in the README! We're currently working with a custom [protobuf repo](https://github.com/uhuruhashimoto/protobufs), meaning you'll need to re-initialize the `protobufs` submodule. I would try running `git submodule sync` and potentially `git submodule update --init --remote` if the first command doesn't work on its own.

Additionally we've added network analysis functionality using a [custom firmware](https://github.com/uhuruhashimoto/firmware), which is why we're using the custom protobufs. Feel free to try it out if you're interested!

D4rk4, 2023-03-22:
@ajmcquilkin problem in this commit - https://github.com/ajmcquilkin/meshtastic-emergency-response-client/commit/082f4a18f7fa73c765a07b9cfd9f0d8002686678#diff-413872f7b4b9114419fdb06b539b6e9c12423e180e770336ab85d068dee7dca5

Why protobufs added in the root of the repo? It's UI only a related commit!

D4rk4, 2023-03-22:
Also build is broken:
```
error: failed to run custom build command for `app v0.1.0 (/home/dark/repos/meshtastic-emergency-response-client/src-tauri)`

Caused by:
  process didn't exit successfully: `/home/dark/repos/meshtastic-emergency-response-client/src-tauri/target/debug/build/app-0c970fddb544ea97/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-changed=protobufs/
  Running: "/home/dark/.cargo/registry/src/github.com-1ecc6299db9ec823/protoc-bin-vendored-linux-x86_64-3.0.0/bin/protoc" "--include_imports" "--include_source_info" "-o" "/tmp/prost-buildcknn8U/prost-descriptor-set" "-I" "protobufs/"
```

ajmcquilkin, 2023-03-22:
> 

Interesting, I hadn't seen the protos get added to the root of the project. I think #355 removes this, I'll make sure in a sec.

Also for your build error, did the protos get loaded correctly? When I've seen that error in the past it's because the protos either don't exist or were broken in some way.

ajmcquilkin, 2023-03-22:
Following up on this, just merged #355 so the top-level `protobufs` submodule should be gone from `main`. Let me know if this errors when `src-tauri/protobufs` is initialized correctly. You can tell if the submodule is correctly initialized if a `NeighborInfo` message exists within the `src-tauri/protobufs/meshtastic/mesh.proto` file.

D4rk4, 2023-05-03:
Working without any issues

closed 2023-05-03 by D4rk4