meshtastic/network-management-client #362: [capability] Not working any more with stock firmaware

D4rk4, 2023-04-11:
After moving to own protobufs, the client stop working with hardware flashed with mainline firmware.
No any info getting from HW. So it's a bug or a feature?
Any way to push your protubfs to upstream? Or backward capability (less feature) mode for the client is possible?
I love your product, but unfortunately, it's can't be used for now in the "official" network.

ajmcquilkin, 2023-04-11:
Interesting, out of curiosity what error are you getting? We've been making some breaking changes, but once [this PR](https://github.com/meshtastic/protobufs/pull/341) (hopefully) goes in we should be able to stabilize the protos mess. Sorry for the chaos, we really appreciate the feedback!

D4rk4, 2023-04-12:
Sorry, it's a hardware issue. All working fine with stock firmware.