meshtastic/network-management-client #368: Can't connect to T-Echo

micheljung, 2023-04-23:
I installed Meshtastic 2.1.9 on a T-Echo device, connected it via serial, it shows up as "COM8" but when I select it:

[image]

```
[2023-04-23T12:16:25Z DEBUG app::ipc::commands] Called connect_to_serial_port command with port "COM8"
[2023-04-23T12:16:25Z DEBUG app::device::state] Set status: Connecting
[2023-04-23T12:16:25Z DEBUG app::device::state] Set status: Configuring
[2023-04-23T12:16:27Z DEBUG app::ipc::events] Dispatching configuration status
[2023-04-23T12:16:27Z INFO app::ipc::helpers] Device configuration timed out, disconnecting device
[2023-04-23T12:16:27Z DEBUG app::device::state] Set status: Disconnected
```

ajmcquilkin, 2023-04-23:
Hi Michel, I'm sorry you're having trouble with the client! I pulled up the client on my Windows (11) device and I'm not able to reproduce this error on versions `2.1.8` or `2.1.10` (most recent dev version at the time of writing). I'm assuming your build flow didn't have any errors given that you're able to run the UI, but one thing I'm wondering is whether the protos version you're on is up to date.

We recently switched from a fork of the Meshtastic protos repo to [using the official repository](https://github.com/meshtastic/protobufs/pull/341), and I'm wondering if this is related. Have you had this repo cloned for longer than a week? If so I would recommend running `git submodule sync --recursive` and then `git submodule update --init --recursive` to make sure the repo is updated with the new protos.

If that doesn't work, would you mind trying the client out on `2.1.10`? I flashed my TBeam directly from within VSCode, I'm not sure if `2.1.10` is out on the web flasher yet.

micheljung, 2023-04-23:
Hi @ajmcquilkin, thank you for your quick response :) I tried 2.1.8 but it doesn't work either. I restarted my computer (to rule out some serial port issues) but still no luck. I wanted to debug it (not being a Rust developer) with IntelliJ but somehow I'm unable to set a breakpoint. Will try another day again.

Meanwhile, let me know if I can provide any further information :)

ajmcquilkin, 2023-04-23:
Yeah of course, thanks for filing the bug! I don't have a T-Echo right now, so my questions are mainly around whether this is a bug with the serial connection on the T-Echo devices or whether this is within this repo. Would you mind validating that serial works on another client (e.g. [web](https://client.meshtastic.org/)) on 2.1.8? In the event that fails I'm happy to poke around in Discord and/or report an issue with serial handling on the T-Echo, but otherwise that might require additional debugging.

Another thought would be whether or not this is a bug that is Windows-only. If you have access to a non-windows machine (macOS, Linux, WSL can't access serial ports so won't work), would you mind trying the client out on a different machine with the same radio? Also while I'm thinking about it, what version of Windows are you working with?

D4rk4, 2023-04-23:
@ajmcquilkin Just FYI, I also try to connect to my NRF-based board from Windows 10 and it's also got the same results, also from WSL and MobaXterm env. Resolved by installing Debian =)

ajmcquilkin, 2023-04-23:
Hmm okay interesting, sounding like the solution is to install Linux then!

@D4rk4 Thanks for the heads-up, seems like this is a Windows-specific problem then. I know that we've had issues with the Windows side of the serial port library we're using in the past, I'm wondering if the same problem is happening here. Did your failure on Windows look fairly similar to the one from this issue? I develop primarily on Linux so until the project gets end-to-end testing up I'll try to validate big changes on Windows 

@micheljung Might not be necessary to try to run the client on a different platform, although if you wouldn't mind still trying your radio on the web client just to make absolutely sure it's not a hardware issue.

I was also thinking a bit more about the debugging process, if you're interested in more detailed debug information coming out of the client, you could change [line 26 in main.rs](https://github.com/ajmcquilkin/meshtastic-network-management-client/blob/c125a0906b9d9e6441c8ee8c70b77a6d8b5985ee/src-tauri/src/main.rs#L26) to use `log::LevelFilter::Trace` instead of the current `Debug` setting. This will activate more detailed output logs, although I'm not positive the logging will catch a serial port error if that's what this is.

D4rk4, 2023-04-23:
> Did your failure on Windows look fairly similar to the one from this issue?
Yes, as I remember it's the same issue.
Funny fact, after I tried to access to meshatstic device port with MNMC (meshtastic-network-management-client), I can't communicate anymore with the device with meshtastic-python, until I reset the MCU. It's always repeatable in Win10, and I never see this behavior in Linux.

ajmcquilkin, 2023-04-23:
That's interesting, thanks for the heads-up! Once we get a minimum reproducible case I'll look into what could be going wrong with the connection flow. A thought that comes to mind is something to do with Windows security, although I don't know if that would explain resetting fixing the problem

micheljung, 2023-04-24:
> Hmm okay interesting, sounding like the solution is to install Linux then!

Not a solution any user wants to hear 😅

I tried with an Heltec v2.0 and it worked. The T-Echo still doesn't. Both work with Meshtastic's web client. As you expected, Trace log didn't produce anything.

I'd be happy to collaborate via Discord to solve this. My ID is Downlord#9855

ajmcquilkin, 2023-04-24:
Only joking! Thanks for bearing with us on figuring out what's going wrong. Good to know that the Windows build is only partially failing to connect, not positive what the difference between the two boards could be. I'll reach out on Discord!

Just for anyone reading this in the future, I'm not imagining this is an architecture related issue. Both the T-Echo and Heltec v2.0 are ESP32 boards, and @D4rk4 experienced the failure on an nRF52 board.

ajmcquilkin, 2023-04-25:
A bit of an update on this, I replaced the [serialport-rs](https://github.com/serialport/serialport-rs) library with the [serial2](https://docs.rs/serial2/latest/serial2/) library just to see if it was a library-specific problem that we're working with, and the problem was exactly the same both times. Both libraries still work with my TBeam fully. Wondering if this is an issue with the driver for the nRF52 module, although I'm not sure in that case why the Python and C# CLIs still can read from the device.

ajmcquilkin, 2023-04-26:
Should be good to go! The issue was primarily based on us not setting [serial DTR](https://en.wikipedia.org/wiki/Data_Terminal_Ready) as with the C# client, but also comes with other general handling improvements.

ajmcquilkin, 2023-04-26:
I'll leave the PR up for validation from @micheljung and @D4rk4 for a bit, but looking to get this merged fairly quickly

D4rk4, 2023-05-02:
@ajmcquilkin Now is working (with my custom NGFF board, based on RAK4630), but after first launch I got 1M old timestamps =)
[image]

ajmcquilkin, 2023-05-02:
Oh fantastic, super glad the hardware is working!!

For the old timestamps, I think that's probably the device keeping really old nodes and the client not filtering them out. Do you have thoughts on how that filtering would work? I'm assuming there would need to be some application-level setting that the user could configure for timeouts, maybe also the option to manually delete connections if the user knows they won't want a given node

D4rk4, 2023-05-02:
@ajmcquilkin relax it's working only once =(
[image]

ajmcquilkin, 2023-05-02:
Oh interesting, what happens if you refresh the port and try a reconnect? I think @micheljung was having a similar issue. Definitely not ideal but hopefully it's workable while a fix is in progress

ajmcquilkin, 2023-05-02:
Hmm okay, so very unreliable but connection is possible. I'll reach out to the Meshtastic devs and see if they've dealt with this before

ajmcquilkin, 2023-05-02:
@D4rk4 Can you pull from `main` and try again? I changed a part of the serial config to better match the C# serial interface

D4rk4, 2023-05-02:
@ajmcquilkin I recheck 5 times with release build, for now it's not reproducible - always successful connect from first try!

ajmcquilkin, 2023-05-02:
Nice, great to hear! I'll re-close this issue, but if this ends up coming back happy to reopen in the future.

PR #369 (2023-04-27): Fix nRF52 connection failure on Windows
This PR fixes a Windows-specific bug where nRF52 devices do not respond to serial commands (RAK, TEcho). This PR takes the following steps to solve the problem and improve the async handling within the Rust backend:

1. Enable [serial DTR](https://en.wikipedia.org/wiki/Data_Terminal_Ready) (thanks @thebentern!)
2.  Switches serial libraries from [serialport](https://github.com/serialport/serialport-rs) to [serial2](https://github.com/de-vri-es/serial2-rs) to enable DTR (see above)
3. Move blocking serial read code into the `spawn_blocking` threadpool to  [... cut]
closed 2023-05-02 by ajmcquilkin