meshtastic/network-management-client #370: Map tiles download via HTTP

D4rk4, 2023-05-03:
Please implement the functionality to change map tile sources to the HTTP protocol. This is essential because HTTP can be effectively cached by various proxy or WAN optimization software, such as Cisco WAAS. This becomes crucial when setting up an off-grid network with limited traffic uplink, as it enables map tiles to be downloaded only once for all devices.

ajmcquilkin, 2023-05-05:
How would you want this to look? Would you be looking for something like an application config field that would allow for you to enter custom URLs for map tile sources?

D4rk4, 2023-05-05:
Thank you for your suggestion to add custom tile server URL configuration to the meshtastic-network-management-client.

Allowing users to provide a configuration string, similar to how it's implemented in LeafletJS, is indeed an excellent idea. For example, users could use the following format for configuring their preferred tile server:

```
https://{s}.google.com/vt/lyrs=s&x={x}&y={y}&z={z}
http://{s}.tile.openstreetmap.org/{z}/{x}/{y}.png
http://tilessputnik.ru/{z}/{x}/{y}.png
```

ajmcquilkin, 2023-05-05:
Pushed a quick PR that adds the option to switch map styles, let me know if that's what you're looking for. Note this won't persist at this point, see #374.

closed 2023-05-06 by ajmcquilkin