meshtastic/network-management-client #373: Replace Remote Splashscreen Image with a Local Image

D4rk4, 2023-05-05:
I have noticed that the current implementation of the meshtastic-network-management-client relies on fetching the splashscreen image from a remote source. This can cause potential issues for users who require offline functionality or have limited network connectivity.

To enhance the offline usability of the application, I recommend replacing the remote splashscreen image with a locally stored image. This change will ensure that the splashscreen is displayed promptly and consistently, regardless of network conditions or connectivity.

Using a local image for the splashscreen will provide the following benefits:

    Improved offline usability, allowing users to access the application's features even when an internet connection is not available.
    Faster load times and reduced dependency on network connectivity, providing a more consistent user experience.
    Enhanced security, as the application will not rely on external sources to fetch the image.

ajmcquilkin, 2023-05-05:
Where are you seeing the application fetching a remote image? Both of the images we use in the onboarding flow are local images, although they do link to Unsplash to credit the photographers. I believe the images are stored in `src/assets` if I'm remembering correctly

D4rk4, 2023-05-05:
Oh, I just grep the code and find usplash links, if it's irrelevant, I close this issue.

closed 2023-05-05 by D4rk4

ajmcquilkin, 2023-05-05:
Oh gotcha, yeah those are attribution links :)

Are you having issues with performance with the images? Happy to open an issue if so