meshtastic/network-management-client #378: Windows 10 x86_64, build from master is broken

D4rk4, 2023-05-05:
[image]
```
C:\Users\User\Desktop>more MNMC.build.CMD
cd C:\Users\User\AppData\Roaming\MobaXterm\home\repos\meshtastic-network-management-client
git pull
git submodule update --init
pnpm i
pnpm run rust:build

C:\Users\User\Desktop>
```

```
vite v4.0.4 building for production...
✓ 2135 modules transformed.
"unix" is not exported by "node_modules/.pnpm/moment@2.29.4/node_modules/moment/dist/moment.js", imported by "src/components/pages/ManageWaypointPage.tsx".
file: C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/src/components/pages/ManageWaypointPage.tsx:4:9
2: import { useSelector } from "react-redux";
3: import type { ColumnDef } from "@tanstack/react-table";
4: import { unix } from "moment";
            ^
5:
6: import type { app_protobufs_Waypoint } from "@bindings/index";
error during build:
RollupError: "unix" is not exported by "node_modules/.pnpm/moment@2.29.4/node_modules/moment/dist/moment.js", imported by "src/components/pages/ManageWaypointPage.tsx".
    at error (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:1967:30)
    at Module.error (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:12962:16)
    at Module.traceVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:13345:29)
    at ModuleScope.findVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:11826:39)
    at ReturnValueScope.findVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:6687:38)
    at ChildScope.findVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:6687:38)
    at ReturnValueScope.findVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:6687:38)
    at ReturnValueScope.findVariable (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:6687:38)
    at Identifier.bind (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:7782:40)
    at CallExpression.bind (file:///C:/Users/User/AppData/Roaming/MobaXterm/home/repos/meshtastic-network-management-client/node_modules/.pnpm/rollup@3.9.1/node_modules/rollup/dist/es/shared/rollup.js:5579:23)
 ELIFECYCLE  Command failed with exit code 1.
       Error beforeBuildCommand `pnpm run ui:build` failed with exit code 1
 ELIFECYCLE  Command failed with exit code 1.
```

ajmcquilkin, 2023-05-05:
Should be good to go, mind pulling and giving it another shot?

D4rk4, 2023-05-05:
Fixed