meshtastic/network-management-client #379: App from latest master not working

D4rk4, 2023-05-06:
The building is fine, but when try to execute the output binary got the error: ```
C:\>C:\Users\User\AppData\Roaming\MobaXterm\home\repos\meshtastic-network-management-client\src-tauri\target\release\meshtastic-network-management-client.exe --port COM4
thread 'main' panicked at 'Logging setup failed: Io(Os { code: 5, kind: PermissionDenied, message: "Access is denied." })', src\main.rs:68:20
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ajmcquilkin, 2023-05-10:
Does this happen on development builds? A prod build on my Windows machine crashed, but I didn't see this error

D4rk4, 2023-05-10:
It's as usual production build.

ajmcquilkin, 2023-05-14:
I haven't had a chance to take a shot at this yet, busy week. I'll try and give it a shot within the next few days. Is this something currently blocking you?

D4rk4, 2023-05-14:
I just grabbed code from the latest master, still hit me.

D4rk4, 2023-05-14:
```
[2023-05-14T20:49:33Z DEBUG app] Logger initialized
[2023-05-14T20:49:33Z INFO app] Building TS types from Rust
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Io(Os { code: 3, kind: NotFound, message: "The system cannot find the path specified." })', src\main.rs:77:77
stack backtrace:
   0:     0x7ff72ae09992 - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ff72ae09992 - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ff72ae09992 - std::sys_common::backtrace::_print_fmt
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\sys_common\backtrace.rs:65
   3:     0x7ff72ae09992 - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\sys_common\backtrace.rs:44
   4:     0x7ff72ae27a7b - core::fmt::write
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\core\src\fmt\mod.rs:1232
   5:     0x7ff72ae050aa - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\io\mod.rs:1684
   6:     0x7ff72ae096db - std::sys_common::backtrace::_print
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\sys_common\backtrace.rs:47
   7:     0x7ff72ae096db - std::sys_common::backtrace::print
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\sys_common\backtrace.rs:34
   8:     0x7ff72ae0bfc9 - std::panicking::default_hook::closure$1
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:271
   9:     0x7ff72ae0bc4b - std::panicking::default_hook
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:290
  10:     0x7ff72ae0c6f8 - std::panicking::rust_panic_with_hook
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:692
  11:     0x7ff72ae0c5ee - std::panicking::begin_panic_handler::closure$0
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:583
  12:     0x7ff72ae0a379 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\sys_common\backtrace.rs:150
  13:     0x7ff72ae0c2a0 - std::panicking::begin_panic_handler
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:579
  14:     0x7ff72ae59cd5 - core::panicking::panic_fmt
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\core\src\panicking.rs:64
  15:     0x7ff72ae5a1f6 - core::result::unwrap_failed
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\core\src\result.rs:1750
  16:     0x7ff72a93bcdd - app::handle_cli_matches::hffdc917f34384910
  17:     0x7ff72aa9cc46 - std::sys_common::backtrace::__rust_begin_short_backtrace::hdfaefca5d6fa8f33
  18:     0x7ff72a98cf1c - std::rt::lang_start::{{closure}}::hde82c7c42d095dbf
  19:     0x7ff72adfe8be - core::ops::function::impls::impl$2::call_once
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\core\src\ops\function.rs:287
  20:     0x7ff72adfe8be - std::panicking::try::do_call
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:487
  21:     0x7ff72adfe8be - std::panicking::try
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:451
  22:     0x7ff72adfe8be - std::panic::catch_unwind
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panic.rs:140
  23:     0x7ff72adfe8be - std::rt::lang_start_internal::closure$2
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\rt.rs:148
  24:     0x7ff72adfe8be - std::panicking::try::do_call
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:487
  25:     0x7ff72adfe8be - std::panicking::try
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panicking.rs:451
  26:     0x7ff72adfe8be - std::panic::catch_unwind
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\panic.rs:140
  27:     0x7ff72adfe8be - std::rt::lang_start_internal
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library\std\src\rt.rs:148
  28:     0x7ff72a93f2dc - main
  29:     0x7ff72ae301cc - invoke_main
                               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
  30:     0x7ff72ae301cc - __scrt_common_main_seh
                               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
  31:     0x7ffdd69d7604 - BaseThreadInitThunk
  32:     0x7ffdd7ce26a1 - RtlUserThreadStart

```

ajmcquilkin, 2023-05-15:
Ah okay I think the issue is related to a relative path not being found when running the application. From `src-tauri/src/main.rs:77`:

```rust
specta::export::ts_with_cfg("../src/bindings/index.ts", &specta_config).unwrap();
```

The client can't find the file since the built project will bundle all the TS into the executable. Not sure exactly how to solve that but I'll poke around and see what I can find.

PR #388 (2023-05-15): Fixed windows build failure due to logging permissions
This PR fixes Windows production builds due to the following:

1. Exporting development types during production builds
2. Logging permissions issues requiring administrator privileges

Closes #379
closed 2023-05-15 by ajmcquilkin