meshtastic/network-management-client #389: Support for TCP connection

micheljung, 2023-05-16:
Would be great if the client couldn't just connect via serial, but also via TCP.

micheljung, 2023-05-23:
I thought there is only an HTTP API but I learned there is also a TCP API. But since HTTP is only supported on ESP32 devices, supporting the TCP API would actually be preferred :)

ajmcquilkin, 2023-05-23:
Interesting, would you mind dropping a link to the TCP API docs? I've only ever worked with the ESP32 HTTP API.

micheljung, 2023-05-24:
https://meshtastic.org/docs/development/device/client-api

https://github.com/meshtastic/python/blob/dc3d43c57cf3dfc95404f5186c4f466ec88036c0/meshtastic/stream_interface.py

https://github.com/meshtastic/python/blob/dc3d43c57cf3dfc95404f5186c4f466ec88036c0/meshtastic/tcp_interface.py

PR #396 (2023-06-26): Add UI for TCP connection flow
This PR builds on work from @micheljung to add UI support for TCP connections. The various UI states for this application are shown below.

Closes #389 
Closes #243 

[image]
_Updated tab-based connection flow_

[image]
_Default screen_

[image]
_Pending connection_

[image]
_Invalid IP_

[image]
_Invalid port on a valid IP_