meshtastic/network-management-client #399: Feature Idea: Location History

jp-bennett, 2023-06-22:
I'm currently using an OwnTracks integration for this, but it would be quite useful to retain location data history. I'm thinking similar to the Google Maps Timeline. Where did this particular radio go in the past 12 hours? On this day last week? etc.

ajmcquilkin, 2023-06-22:
That sounds like a cool feature! Would you mind attaching some design inspiration? I'm not familiar with that in Google Maps

jp-bennett, 2023-06-22:
[image]
[image]

ajmcquilkin, 2023-06-22:
Thanks! I'll start playing around with some designs and see how this could fit into the mapping interface. How would you expect to open the timeline view for a given node? I'm imagining that clicking on a node could open a menu which could have an option to view its history? Super open to thoughts on this one

jp-bennett, 2023-06-23:
There might be an option to show "tails" on the normal map view, to see where each node has immediately been. 

And then when on the map view, maybe add the option to view a device's movements in the device details box. The only thing that comes to mind there is that finding and clicking on an individual node might be fiddly. So maybe list of nodes with location data on the far left, as a submenu of "View Map"