meshtastic/network-management-client #419: Mange Nodes location shows Latitude, Longitude as  -0.00000°,  0.00000° whilst fixed location configured for nodes

markbirss, 2023-07-05:
meshtastic-network-management-client_0.2.0_amd64.AppImage

[image]

ajmcquilkin, 2023-07-10:
Would you mind giving some information on how you've set up your nodes? I'm setting a fixed position with the Python CLI and the position is showing up on the map.

markbirss, 2023-07-10:
@ajmcquilkin 

Sure, i set via CLI with
`meshtastic --port /dev/ttyACM2 --set position.fixed_position true --setlat -lat --setlon lon`

They show position on map correctly it just the "Manage Node" view showing as -0.00000°, 0.00000°

ajmcquilkin, 2023-07-10:
Ah sorry, I had misunderstood your issue. Found the bug; I had forgotten to remove something from a recent refactor. The fix is pushed to `main` and will be shipped in the next release

markbirss, 2023-07-10:
Ok to close as completed.

closed 2023-07-10 by ajmcquilkin