meshtastic/network-management-client #435: Run Rust:Dev error - Failed to get Cargo Metadata

ayysasha, 2023-08-19:
[image]

After pulling down the repo and running installation instructions, I am being met with the following after running >pnpm run rust:dev in the root directory of the repository.

`       Error failed to get project out directory: failed to get cargo metadata: No such file or directory (os error 2)
 ELIFECYCLE  Command failed with exit code 1.
`

ajmcquilkin, 2023-08-19:
I'm sorry to hear you're having issues! Just for reference, what version of Rust do you have installed? My next thought would be whether or not you cloned the submodules. I've not seen this specific error but not having the protocol buffer files installed could do something like this

ayysasha, 2023-08-19:
An update of my rust environment fixed this, ill make PR to add that to the installation steps :)

Thank you!

ajmcquilkin, 2023-08-19:
Glad to hear it, thanks for the PR!

Tomi05-k, 2023-08-31:
I also had the error at first. 
I had installed RUST and the other packages first.
You should reboot after the installation.

then it runs.

(Use MacOS)

Many greetings Tom!

closed 2024-03-07 by ajmcquilkin