meshtastic/network-management-client #439: Windows 10 x64, production build failed

D4rk4, 2023-09-16:
`git fetch
git checkout main
git reset --hard FETCH_HEAD --recurse-submodules 
git submodule update --init
pnpm i
pnpm run rust:build`

Multiple errors like
`src/components/config/channel/ChannelConfigDetail.tsx:154:13 - error TS2322: Type 'string | FieldError | Merge<FieldError, FieldErrorsImpl<any>> | undefined' is not assignable to type 'string | undefined'.

154             error={errors.name?.message}
                ~~~~~

  src/components/config/ConfigInput.tsx:13:3
    13   error?: string;
         ~~~~~
    The expected type comes from property 'error' which is declared here on type 'IntrinsicAttributes & Pick<IConfigInputProps & UseFormRegisterReturn<string>, "style" | "color" | "size" | ... 288 more ... | "step"> & RefAttributes<...>'

src/components/config/channel/ChannelConfigDetail.tsx:170:13 - error TS2322: Type 'string | FieldError | Merge<FieldError, FieldErrorsImpl<any>> | undefined' is not assignable to type 'string | undefined'.

170             error={errors.uplinkEnabled?.message}
                ~~~~~

  src/components/config/ConfigInput.tsx:13:3
    13   error?: string;
         ~~~~~
    The expected type comes from property 'error' which is declared here on type 'IntrinsicAttributes & Pick<IConfigInputProps & UseFormRegisterReturn<string>, "style" | "color" | "size" | ... 288 more ... | "step"> & RefAttributes<...>'

src/components/config/channel/ChannelConfigDetail.tsx:178:13 - error TS2322: Type 'string | FieldError | Merge<FieldError, FieldErrorsImpl<any>> | undefined' is not assignable to type 'string | undefined'.

178             error={errors.downlinkEnabled?.message}
                ~~~~~

  src/components/config/ConfigInput.tsx:13:3`

ajmcquilkin, 2024-01-07:
This should be fixed in #443, thanks for the heads-up. Will let you close the issue when the PR goes through.

ajmcquilkin, 2024-03-07:
#443 is merged, should be fixed.

closed 2024-03-07 by ajmcquilkin