meshtastic/network-management-client #444: Client stops decoding packets with "Failed to send decoded packet: channel closed"

KodinLanewave, 2024-01-12:
After a while, the client stops decoding updates and sends the following to console:

```shell
called `Option::unwrap()` on a `None` value
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[2024-01-12T00:08:56Z ERROR meshtastic::connections::stream_buffer] Failed to send decoded packet: channel closed
[2024-01-12T00:09:00Z ERROR meshtastic::connections::stream_buffer] Failed to send decoded packet: channel closed
[2024-01-12T00:09:11Z ERROR meshtastic::connections::stream_buffer] Failed to send decoded packet: channel closed
[2024-01-12T00:09:17Z ERROR meshtastic::connections::stream_buffer] Failed to send decoded packet: channel closed
[2024-01-12T00:09:45Z ERROR meshtastic::connections::stream_buffer] Failed to send decoded packet: channel closed
``` 

This is when connected to a t-beam via wifi. (serial doesn't work at all as the wrong dev list is populated on Ubuntu)

ajmcquilkin, 2024-01-12:
Interesting, thanks for the report! How long does it take (approximately) for this to happen?

KodinLanewave, 2024-01-12:
@ajmcquilkin Approximately 5-10 minutes or so. Somewhat random.