meshtastic/network-management-client #445: serial port not working on Arch Linux

cyber-vi-king, 2024-01-25:
When I run `pnpm run rust:dev` the client shows me two serial ports, but not `/dev/ttyUSB0` and `/dev/ttyACM0` (that's the RAK) but instead `/sys/class/tty/ttyACM0` and `/sys/class/tty/ttyUSB0` both of which claim "is a directory" and don't work. Is there anything I can fix or any way to pass the right port as parameter on running `pnpm run rust:dev`?

marcoSanti, 2024-02-05:
Same thing on Ubuntu 23.10

matthewCmatt, 2025-02-17:
Ran into this on Ubuntu 22.04 a bit back.

I think I solved this by making sure I have permission to access the serial port. Something like:
```sh
ls -l /dev/ttyUSB0 &
sudo usermod -aG dialout $USER
```

JonathanMH, 2026-08-16:
on arch you have to:

```
sudo usermod -a -G uucp $USER
```
and reboot

source: https://forum.manjaro.org/t/usermod-group-dialout-does-not-exist/106523