meshtastic/network-management-client #453: Connection timed out: Connecting to meshtasticd on Linux

xstrex, 2024-02-27:
Running a Raspberry Pi 5, with Waveshare LoRa hat. `meshtasticd` service is up and running, and talking to radio. 

Running MNMC 0.3 on a Mac.

When running MNMC and choosing TCP, put in the IP of the PI & 4403 for the port. Hit connect, and receive the following form the system journal:
```
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ApiServer] Incoming wifi connection
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] Client wants config, nonce=3712423828
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] Starting API client config
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] getFromRadio=STATE_SEND_MY_INFO
Feb 26 18:46:59 DietPi meshtasticd[723]: DEBUG | ??:??:?? 15508 [ServerAPI] encoding toPhone packet to phone variant=3, 12 bytes
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] getFromRadio=STATE_SEND_METADATA
Feb 26 18:46:59 DietPi meshtasticd[723]: DEBUG | ??:??:?? 15508 [ServerAPI] encoding toPhone packet to phone variant=13, 29 bytes
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
Feb 26 18:46:59 DietPi meshtasticd[723]: INFO  | ??:??:?? 15508 [ServerAPI] nodeinfo: num=0xddad002e, lastseen=0, id=!ddad002e, name=Tnix Router
```
But the MNMC client times out, and never connects. 

If I telnet to the same IP & port it connects (but obviously doesn't do anything)
```
telnet 192.168.2.110 4403
Trying 192.168.2.110...
Connected to 192.168.2.110.
Escape character is '^]'.
```

If I were to guess the client isn't waiting long enough to connect to the service, or not receiving the correct information from the service. 

Happy to provide logs, or anything necessary. Thanks.

ajmcquilkin, 2024-03-07:
This looks like a config timeout issue, where it takes longer to connect to a TCP node than a serial node. #456 bumps this timing to 15s, although at some point I want to make this user-configurable.

xstrex, 2024-08-22:
It's been 6mo, any progress on this?

garthvh, 2024-08-23:
> It's been 6mo, any progress on this?

You are welcome to dig in

xstrex, 2024-10-12:
> > It's been 6mo, any progress on this?
> 
> 
> 
> You are welcome to dig in

Not opposed to digging in, also not entirely sure where to start, some pointers or coding standards might be helpful.