meshtastic/network-management-client #454: Dynamic heatmap showing coverage of a specific network

lyrics1988123, 2024-02-29:
we (friends of me and myself) had an idea for a heatmap which shows the range of every node.
for wifi there are apps that achieve the same, but for meshtastic i was not able to find something like that - so i got asked by AJMcQuilkin (Meshtastic Discord) to create a Issue/Feature Request here :)

i thought about it like that way:
- using the neighbor info feature to send node names, their gps location data and rssi via mqtt to the network-management-client
- every node should provide those information in like once per hour (user specified for "wardriving tests" to map the network)
- the NMC uses the gathered data to create a map based on OSM or something to show the following things:

1. where a signal was able to get picked up (heatmap style)
2. which of the nodes are connected to each other (lora of course, not mqtt)

Im not sure if this really is doable the way im thinking of it, but i think about coverage map from LTE prodiders and stuff.
The Main reason for all that is, to visualize the actual reach and coverage of the network and learn where the limitations are. 
Im can see a usecase for me: i would plate a node on my cars roof and drive around and the software creates the heatmap, afterwards i can see which places in my village are "deadzones" and which places are already covered by the already existing nodes in the area. 

i could also see a case where you would have an official map everyone can use which lets you see existing networks, this way i could see that my ~40 nodes big network is not the only one in the area but that the other one is using different settings --> potential for binding them together if contact data are provided to the network name on the map. not sure how to decide how or when a network should be published to such a public map, but could be a great tool.

im pretty sure there is some more brainwork to do, but technically the neighbor info, mqtt and Network-Management-Client should together have everything needed.

thanks!

asjmcguire, 2024-03-04:
When the "hopsaway" firmware arrives - then every packet will be tagged with how many hops it travelled to reach your node, this will make dynamically figuring out a coverage circle much easier, and does not reply on people enabling the neighbourinfo module which significantly impacts the mesh.

lyrics1988123, 2024-03-06:
> When the "hopsaway" firmware arrives - then every packet will be tagged with how many hops it travelled to reach your node, this will make dynamically figuring out a coverage circle much easier, and does not rely on people enabling the neighbourinfo module which significantly impacts the mesh.

interesting, that would be a cleaner solution, i like that.

matthewCmatt, 2025-03-03:
This functionality is somewhat covered by the [Meshtastic Site Planner app](https://meshtastic.org/docs/software/site-planner/)