meshtastic/network-management-client #470: [Bug]: error[E0282]

John-JacobJingleHeimer-Schmidt, 2024-08-22:
## Describe the bug

Compiling error in MacOS

## To Reproduce

1. run pnpm run rust:dev
Response:
error[E0282]: type annotations needed for `Box<_>`
 --> /Users/REDACTED/.cargo/registry/src/index.crates.io-6f17d22bba15001f/time-0.3.34/src/format_description/parse/mod.rs:83:9
   |
83 |     let items = format_items
   |         ^^^^^
...
86 |     Ok(items.into())
   |              ---- type must be known at this point
   |
help: consider giving `items` an explicit type, where the placeholders `_` are specified
   |
83 |     let items: Box<_> = format_items
   |              ++++++++

## Expected behavior

Should compile app

## Screenshots

If applicable, add screenshots to help explain your problem.

## Machine Information

 - OS: [e.g., Ubuntu]
 - Version: [e.g., 23.04 LTS]

## Additional context

Add any other context about the problem here.

jamesarich, 2024-08-25:
fwiw i was able to get past this by deleting `src-tauri/Cargo.lock` and trying again.

John-JacobJingleHeimer-Schmidt, 2024-08-25:
> fwiw i was able to get past this by deleting `src-tauri/Cargo.lock` and trying again.

That did the trick, thanks!

matthewCmatt, 2025-05-12:
The Cargo.lock has since been regenerated so I'm going to mark this as solved

closed 2025-05-12 by matthewCmatt