meshtastic/network-management-client #471: [Bug]: pnpm run rust:dev fails with `Found argument '/usr/bin/node-20' which wasn't expected, or isn't valid in this context`

alistair23, 2024-10-12:
## Describe the bug

I have followed the steps from https://github.com/meshtastic/network-management-client?tab=readme-ov-file#installation for Fedora 40, but when running `pnpm run rust:dev` I get the following error

```shell
$ pnpm run rust:dev

> meshtastic-network-management-client@0.3.1 rust:dev /var/home/alistair/sources/network-management-client
> tauri dev -- --no-default-features --

error: Found argument '/usr/bin/node-20' which wasn't expected, or isn't valid in this context

USAGE:
    cargo-tauri [OPTIONS] <SUBCOMMAND>

For more information try --help
```

## To Reproduce

1. Try to build following steps from https://github.com/meshtastic/network-management-client?tab=readme-ov-file#installation

## Expected behavior

Successfully build and run the application

## Screenshots

If applicable, add screenshots to help explain your problem.

## Machine Information

 - OS: Fedora
 - Version: 40

## Additional context

Add any other context about the problem here.

alistair23, 2024-10-16:
The file does exist

```shell
$ ls -al /usr/bin/node-20
-rwxr-xr-x. 1 root root 28176 Sep 11 10:00 /usr/bin/node-20
```

matthewCmatt, 2025-02-18:
A few debugging ideas:
- Try upgrading to Node 22?
- Are you able to run `pnpm i`? `pnpm ui:dev`? `pnpm ui:lint`? `pnpm tauri`?
- Could you drop your versions from `node -v`? `pnpm -v`? `cargo version`?

matthewCmatt, 2025-07-03:
Closing as this is stale and can't reproduce.

Feel free to reopen if you still see this with latest node+pnpm+cargo versions

closed 2025-07-03 by matthewCmatt