meshtastic/network-management-client #472: [Feature Request]: Docker support

Th3H0ff, 2024-10-13:
I managed to put together a Dockerfile that does compile but stops on some Tauri error. I haven't run across Tauti before and have a tough time proceeding further. I did some troubleshooting with the help of ChatGPT but got stuck on some Tauri configuration files .

The Dockerfile I've created

```
FROM node:18-slim

# Install system dependencies required by Tauri and Rust
RUN apt-get update && \
    apt-get install -y curl build-essential libssl-dev libwebkit2gtk-4.0-dev libgtk-3-dev \
    libayatana-appindicator3-dev librsvg2-dev && \
    rm -rf /var/lib/apt/lists/*

# Install Rust and Cargo
RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
ENV PATH="/root/.cargo/bin:${PATH}"

# Install the Tauri CLI using Cargo
RUN cargo install tauri-cli && \
    echo "Tauri installed, checking version..." && \
    ls /root/.cargo/bin && \
    /root/.cargo/bin/cargo-tauri --version

# Install PNPM globally
RUN npm install -g pnpm

# Set the working directory in the container
WORKDIR /usr/src/app

# Copy the package.json and pnpm-lock.yaml to the container
COPY package.json pnpm-lock.yaml ./

# Install dependencies using PNPM
RUN pnpm install

# Copy the rest of the application code to the container
COPY . .

# Build the frontend using Vite
RUN pnpm run ui:build

# Expose the port on which the application will run
EXPOSE 5173

# Set the environment variable for running in development mode
ENV NODE_ENV=development

# Command to run the Rust development server using PNPM
CMD ["cargo-tauri", "dev", "--", "--no-default-features"]
```

matthewCmatt, 2025-02-18:
Hey @Th3H0ff, could you explain a bit more about this suggestion? Are you thinking about dev containers?

If you're looking for a Meshtastic client to be hosted from a container, consider [the web client](https://github.com/meshtastic/web). If you want to run Meshtastic client commands from inside a docker container in a more automated environment, [the Python library/CLI](https://github.com/meshtastic/python) would probably be easier

I'm not sure if the above have already have container files made though. Would be a great contribution if not!

matthewCmatt, 2025-05-12:
Partially addressed by #493. Could use a standalone Dockerfile for non-VSCode development or CI.