meshtastic/network-management-client #474: [Bug]: Node shut downs 5 seconds after serial connected

RadioactivePotato, 2024-12-29:
## Describe the bug

The node shuts down after 5 seconds when connected via serial on version v0.3.0, tried on 2 different device/os, same result

## To Reproduce

1. Connect to node via serial
2. Look at the screen on the node

## Expected behavior

The node shouldn't shut down

## Screenshots

N/A

## Machine Information

 - OS: Windows 10 22H2 (OS Build 19045.3803) (.msi installer), Manjaro Linux (Appimage)

## Additional context

Add any other context about the problem here.

matthewCmatt, 2025-02-07:
Could you share information on the node you were running? What hardware? What Meshtastic version was it running?

Thanks for the report!

RadioactivePotato, 2025-02-09:
Heltec v3.2
2.5.20.4c97351

W1BTR, 2025-04-12:
Same issue here. Heltec V, firmware 2.6.4. It starts up fine, then when I launch the app it prompts me for a serial port. I can select it, and the app continues, but the heltek is stuck on "Shutting Down..." indefinitely until I close the app.

W1BTR, 2025-04-12:
Only was able to get it working via bluetooth and wifi, no direct serial communication worked.

matthewCmatt, 2025-05-06:
I also was able to reproduce this just by getting a Heltec v3 connected on firmware 2.6.4

sturzl, 2025-05-11:
Same for me.

Steps:
- Install .deb
- `missing libssl1.1 dependency`
- install that
- click on `/dev/ttyUSB0`
- Heltec screen stuck on text `Shutting down...`
- App shows a connection error

App Version: [app-v0.3.0](https://github.com/meshtastic/network-management-client/releases/tag/app-v0.3.0)
OS: linux mint 6 (Debian 12.1)
Device: Heltec V3 board v3.2
Firmware: [Meshtastic v2.6.4.b89355f](https://github.com/meshtastic/firmware/releases/tag/v2.6.4.b89355f)

[image]

[image]

RadioactivePotato, 2025-05-11:
Does your computer have permission to access the device?

try running

```
$ sudo adduser <username> dialout
$ sudo chmod a+rw /dev/ttyUSB0
```

sturzl, 2025-05-11:
I was in the dialout group, but hadn't added the permissions to `/dev/ttyUSB0`. But after updating that I still have the same result. I also tried running everything as root and again a connection timeout error :/

[image]

nfacha, 2025-06-06:
Can confirm also facing this issue on the same circunstances

RadioactivePotato, 2025-09-18:
Is anyone experiencing this issue on other esp32 based devices that isn't the heltec v3?