meshtastic/network-management-client #498: [Bug]: All nodes ever apparently have a direct link to each other?

W1BTR, 2025-04-12:
## Describe the bug

A clear and concise description of what the bug is.

## To Reproduce

1. connect a heltek v3 (not sure if this works on other radios)
2. wait for a mesh map to show
3. Every node that comes up is shown to have a direct path to all other nodes

## Expected behavior

I expect to see a map of nodes and a display of what paths they have to each other, or an option to hide paths

## Screenshots

[image]
If applicable, add screenshots to help explain your problem.

## Machine Information

 - OS: Windows 11

## Additional context

I have a little node with a compromise antenna that just needs to make it to a repeater down the road. There is no way I have a direct path to any nodes other than that. This is also reflected in the android meshtastic app, in which I dont see as many nodes and there is only one device that is 1 hop away.

matthewCmatt, 2025-04-13:
I have definitely seen this in the latest release. I think these link lines were removed/disabled since then, so this should be "solved" by the next release.

W1BTR, 2025-04-15:
Noted, thank you. Ill take a look come next release

closed 2025-07-03 by matthewCmatt